KenT2/pipresents-beep #3: OMX Player Error

SphinxJ, 2019-01-04:
Every now and then my pi's are getting the following error:

"Profile Error: omxplayer ended before loading track"

I am using the liveshow playlist. I have increased the memory on all the pi's to 256MB as recommended in the setup files. I have not went through any log files yet. This error has happened several times since deploying three pi's to the field. Happens maybe once a day. Any ideas on what could be causing this? What information can I obtain to help find the solution to the problem?

Older PP version issues:
https://github.com/KenT2/pipresents-gapless/issues/1
https://github.com/KenT2/pipresents-gapless/issues/13
https://github.com/KenT2/pipresents-gapless/issues/61

KenT2, 2019-01-10:
This has happened with a few users, it seems to be connected to home made videos so it might be worth rendering the video using a different program.
I have failed to  re-create the problem here with later versions of PP. If you can let me have the video it would be useful. Contact me via the PP Wordpress blog and I will send an email address

SphinxJ, 2019-01-10:
Thanks Ken. I will send you an email.

SphinxJ, 2019-01-16:
Sent you an email through your website and included a link to my dropbox which has all the videos I am playing on our pis. Thanks for looking into this.

KenT2, 2019-02-16:
To let the world know where we are.
 I noticed from your logs that you were reading the videos from a server over a business network and this was producing a number of errors for which I have the logs. The cause of your problem is not clear and I  will investigate further.
You moved the videos so they were resident on the Pi and the problem has gone away. Whether it is errors  in your server or just delays caused by a busy server/network remains to be seen.

I have been unable to reproduce the problem here when running you videos over a home network.

KenT2, 2019-04-17:
Update:
I have had one failure when using  my home network, after 19 days. The video terminated early.

I looked at your and my logs. I can see no clear pattern. The most prevalent report was that a video terminated early which sounds indicative of breaks/gaps in feed from the video source. The other failure was failure of the dbus connection to omxplayer. This is not directly related to the network and could be a result of omxplayer dying prematurely rather than a dbus failure.

It would be worth looking at or raising an omxplayer issue on its github to see if altering an of its command line options might help the situation.

SphinxJ, 2019-04-18:
Thanks for the update Ken.

Since moving my files to the local storage drive on the PI, I have not seen any issues.

closed 2019-04-27 by KenT2