KenT2/pipresents-beep #6: Transition Screen for RadioButtonShow

SphinxJ, 2019-04-23:
Ken,

I am finally getting around to converting my liveshow to a radiobuttonshow. Using your example, I was able to create a radio button show that plays several videos in a loop state as the default. I have two push buttons attached to the GPIO pins, where each one controls a unique single video (subshow). I am able to press either button in the middle of whatever is currently playing and once the video is finished, the default looped videos begin playing. So for the most part, everything is working right now as I would have hoped it would.

My current issue is the transition between the default looped video and the subshow video. When I press either button, the screen flashes back to the desktop with a windowed version of the PP shown instead of staying in fullscreen mode. Once the subshow video is pulled up, it goes back to full screen and plays that video without any issues. After that subshow video finishes, the PP returns to the default looped videos without flashing back to the desktop.

Is this how it is suppose to function? Is there anything I can do to smooth out the transition so the desktop is not seen?

Thanks for the guidance in the past to move to the radio button show!

SphinxJ, 2019-04-23:
Just a quick update.

I just did a reboot and it seems to be working fine now. The screen does go black between videos, which is just what I wanted.

Side note, I was manually calling up the profile I created instead of how I normally run the program which is through an autostart command. This may have been the issue all along. Now to final wiring and installing the buttons on the TV.

Thanks again for all the hard work you put into the development of PiPresents! Excellent coding!

KenT2, 2019-04-23:
Use the -f command line option. The gap should then be black or whatever background colour is set in the Start Show

SphinxJ, 2019-04-23:
Thanks. I throw -f -b at the end of the startup routine, but I did not use that when I manually called the profile.

closed 2019-04-25 by SphinxJ