KenT2/pipresents-beep #7: Going to track by inputted number

hmnd, 2019-04-30:
How would I go about implementing a way to go to a given track by inputting its number into a number pad? I know one way would be to set up an individual event handler for every track, but I'm hoping there's a simpler solution, perhaps?

KenT2, 2019-04-30:
Use a radiobuttonshow with controls like play1 play track 1, play2 play track2 The example pp_radiobuttonshow_1p4 does this using the autogenerated pp-key-1 pp-key-2 events to respond to normal keyboard keys.

If the numberpad looks like a standard keyboard then you may get away with the autogenerated pp-key-1 etc. or by modifying keys,cfg to configure the standard keyboard driver pp_kbddriver.py to produce the event play1 and play2 . Otherwise you may need to use another I/O plugin (e.g.serialdriver) or write one yourself.

hmnd, 2019-04-30:
Thanks for the fast reply! pp-key works with the numpad, however, how would I go about making it able to handle double digits?

KenT2, 2019-04-30:
You woudl  need to modify pp_kbddriver.py to read characters and form them into a string which is then converted to an event. The input would need to be terminated by enter. IIRC pp_serialdriver.py allows strings

KenT2, 2019-08-07:
pp_kbddriver_plus.py should be able to do this. Not sure why I did not say it earlier :-(

closed 2019-08-07 by KenT2