KenT2/pipresents-beep #8: Statistics Production

HoneyBearUK, 2019-05-05:
Hi Ken,
I'm trying to use the stats to record the amount of times a Show is selected, controlled by a GPIO button call.
I'm using PP 1.4.1a and i'm including the '-d 128'.
In the output file i'm only getting entries indicating when PP was started although there is activity on the Terminal Screen but no visible text.

What have i missed, should i be adding other commands/calls in?

We have discussed this feature in the past but I have only just come to build it in to my program.

Many thanks,
Stewart

KenT2, 2019-05-05:
You seem to be doing all the right screens. Are you looking in pp_stats.txt
Which type of show are you using and what do you mean by selecting a show
Beep is now at 1.4.2a which I issued a couple of weeks ago with enhanced stats recording. I know that worked.

HoneyBearUK, 2019-05-05:
Hi,
I'm looking in pp_stats.txt
I have now upgraded to 1.4.2a, updated the profiles and still get the same results.
This is the text for when i started PP -
"2019-05-05";"23:29:45";"";"";"";"start";"";"";"";"zoo2019_v2";"zoo2019_v2"

1 minute later 23:30:xx i started a Mediashow (show_1) but no additional lines printed to the log.
show_1 is configured in pp_io_config/gpio.cfg as a GPIO and in pp_io_config/keys.cfg as a number key. 
When i start PP with the -d 128 switch the cursor in the terminal screen moves down 9 lines but nothing is shown and when i then start the show it moves down a further 5 lines with nothing shown.

Back in Nov 2018 this worked OK - Issue #92

Stewart

KenT2, 2019-05-08:
The reason is that I do not record a stats record when opening a show. I will put that in the next version. Meanwhile you can edit pp_show.py add after line 156

        self.mon.stats(self.show_params['type'],self.show_params['show-ref'],self.show_params['title'],'open show','','','','')   

using leading spaces to line it up with the previous if statement. You can use Leafpad the text editor to do this

HoneyBearUK, 2019-05-08:
Hi Ken,
I've tried that, no difference, no additional output (lines) pp_stats.txt

"2019-05-08";"20:43:52";"";"";"";"start";"";"";"";"zoo2019_v2";"zoo2019_v2"
"2019-05-08";"21:12:20";"";"";"";"start";"";"";"";"zoo2019_v2";"zoo2019_v2"
"2019-05-08";"21:15:28";"";"";"";"start";"";"";"";"zoo2019_v2";"zoo2019_v2"

I only get an entry when PP is started.

Entry made in pp_show.py

          self.end('error',message)
            self.mon.stats(self.show_params['type'],self.show_params['show-ref'],self.show_params['title'],'open show','','','','')

    # dummy, must be overidden by derived class

Many thanks for your ongoing support.
Stewart

HoneyBearUK, 2019-05-08:
Hi Ken,

 

I have posted a reply on Git Hub but so far no difference.

 

The way I’m reading the IF statement is 

 

“ If reason == ‘error’ “  the next 3 lines will run including the line I have added.

 

This will need an error to work, but is the action of selecting a show an Error?

 

Regards,

Stewart

KenT2, 2019-05-08:
Sorry my bad wording, It needs to line up with the preceding if  . You probably need to remove 4 spaces

       if.....
           blah, blah
       self.mon.stats(......

HoneyBearUK, 2019-05-08:
Hi,

 

Many thanks, that works 😊

 

I will close the case on Git Hub.

 

Stewart

closed 2019-05-08 by HoneyBearUK