KenT2/pipresents-beep #11: Make radiobuttonshow etc. 'gapless'

KenT2, 2019-10-08:
Ken, I have now done this so it runs full screen and disabled blanking and everything works perfectly except while my video screensaver is running and i select a video, it drops out to the image background for about a half second before while it loads the video. Is there any way to have the screensaver video freeze on the it's current frame or have that video continue until the newly selected video begins playing? Also if I trigger a video while another is currently playing I get the same behavior of it dropping to a background image for a second where ideally it would freeze on a frame to be truly gapless. If I launch a video and it plays to completion, it will freeze on the last frame and be gapless while loading back to the screensaver. 

This ia a limitation of PP. In a mediashow I disguise the gap by pausing the video near its end while the next video loads. With a radiobuttonshow I don't do this. However your post has given me an idea of how it might be done. Meanwhile the fudge below might work:

For the gpio's that start the videos use the press edge to send the pause command and the trailing edge to send the play command. My proper fix would do achieve the same by pausing the video in the play command code.

volcanicClay, 2019-10-09:
Ken, is this a command I would execute in the GUI?  Or do I need to edit the GPIO config file?

KenT2, 2021-07-05:
Going from First track to another track 'freeze at end' works to cover the gap due to loading.
However I suspect your screensaver is a MediaShow. Unfortunately freeze at end does not cover the gap when one show exits to another. Would be very complicated to achieve this.

closed 2021-07-05 by KenT2