KenT2/pipresents-beep #12: Sync videos betwee multiple RPi's

KenT2, 2019-10-27:
I came accross follwing link to sync the videos on multiple rpis,
it works well for me on 9 nos of rpi without any problem
if I want to integarte that in pi presents can you tell me how to do it.
https://github.com/turingmachine/omxplayer-sync

Transferred from gapless  [https://github.com/KenT2/pipresents-gapless/issues/117](url)

ninadk1092, 2019-10-31:
@KenT2 any estimate when this issue will be fixed and released?

KenT2, 2019-11-04:
maybe this winter

RogierQ, 2020-02-16:
o nice! is it working on gapless?
as winter is almost gone ;)

KenT2, 2020-02-16:
Ha Ha, winter is far from over, no snow yet.
It will never be in Gapless, only in Beep.
Got somewhat diverted because the Foundation announced there would be no further development on omxplayer and we should use VLC. Tried VLC but its some way from being an embedded videoplayer. Now I'm trying to sort out rotation of touchscrreens
Its going to be quite a project to do it so don't hold your breath, it may not even be possible.

closed 2021-07-04 by KenT2

KenT2, 2021-07-04:
omxplayer is now deprecated and VLC has replaced omxplayer in Pi Presents. Not sure VLC can sync videos between multiple Pi's

pramodmadhav, 2022-02-23:
can one set position in VLC (hr:min:sec:frames)

KenT2, 2022-02-24:
You can in VLC but not in Pi Presents.