KenT2/pipresents-beep #14: Fade between slides

CooperWhitlow, 2020-01-17:
I don't see fade transitions between slides/images as a feature. Can you provide any guidance on where to start implementing this myself?

KenT2, 2020-01-19:
You would need to modify pp_imageplayer.py. I tried fading when I first developed PP using a first generation Pi but it was far too slow as the CPU has to alter the alpha of every pixel n times a second. Might work with Pi4 though as it's 10 or 20 times faster. show() is called when the image is first displayed and hide() when it is removed.

A better solution would be to use opengles which would allow all sorts of transitions. I tried a demo here but struggled to get it working.  [https://github.com/KenT2/ppi3d](url)

KenT2, 2021-03-07:
I tried fades with a Pi4, still too slow.

closed 2021-03-07 by KenT2