KenT2/pipresents-beep #21: Image window without grey/yellow thinny border in no full screen mode.

francovico, 2020-08-19:
Hi K
[image]
en.
First of all I want to congratulate you for making such excellent software and sharing it openly with us. Just 1 week ago I am starting to test and use it and it is incredible.

I have a Raspberry Pi4 that I manage remotely with VNC.
In the HDMI1 output (it would be the second output physical) I have a screen connected.
In the HDMI0 output I have nothing, but it is forced to work from the boot.txt.
I don't want to use the software in fullscreen because I lose VNC remote management of the device on the first screen.
Modify the openbox files according to your suggestions with <decor> no </decor>, and also other modifications on my own like border = no, border pix = 0, etc.
I can't remove the small gray or yellow border from the window!
Edit all the theme files that I found. I don't know more to try.
Is there a way to achieve this without having to use it in Full Screen so as not to lose remote control of the device?
Its only appears in images, not in video.

[image]
[image]

I hope I have explained myself well.

Thank you.
Regards.

Franco from Argentina.

KenT2, 2020-08-19:
>I have a Raspberry Pi4 that I manage remotely with VNC.
>In the HDMI1 output (it would be the second output physical) I have a screen connected.
>In the HDMI0 output I have nothing, but it is forced to work from the boot.txt.

Not sure why you need to do this, it may be this that is confusing PP and producing the yellow lines.
 
>I don't want to use the software in fullscreen because I lose VNC remote management of the device on the first screen.

On a Pi4 the Raspbian desktop spans the size of the two monitors (assuming that they are configured side by side in the screen configuration utlitiy). VNC puts this 2 monitor wide desktop in a single window. If you scroll the VNC window sideways you will see the second monitor screen.

PP must use the --fullscreen option when running to get rid of the borders and title bar. However without --fullscreen when running  it reverts to normal windowed mode for HDMI0 but not for HDMI1 which remains fullscreen.

>Modify the openbox files according to your suggestions with no , and also other modifications on my own like border = no, border pix = 0, etc.
>I can't remove the small gray or yellow border from the window!
>Edit all the theme files that I found. I don't know more to try.
>Is there a way to achieve this without having to use it in Full Screen so as not to lose remote control of the device?

You can disable the yellow lines. In pp_displaymanager.py line 591 set thickness to 0

>Its only appears in images, not in video.
Yes, display of video is different to images. It is in a layer in front of the images so hides the lines and the image.

francovico, 2020-08-19:
Dear Ken,

Thank you!

It works for me:
**>  You can disable the yellow lines. In pp_displaymanager.py line 591 set thickness to 0**

> PP must use the --fullscreen option when running to get rid of the borders and title bar. However without --fullscreen when running it reverts to normal windowed mode for HDMI0 but not for HDMI1 which remains fullscreen.

This doesn't work for me like that. My desktops in VNC look the way you say: 2 monitor wide desktop in a single window. But when I run it in -f full screen, the whole desktop of screen 0 goes black without being able to use it. I tried to restore it anyway but was unsuccessful.
Maybe it is some problem in my Raspbian, but I mention it so that you can check it. I will try to reinstall everything.

Thank you again.
**It's solved.**
Regards.

Franco

closed 2021-03-07 by KenT2