KenT2/pipresents-beep #22: keys.cfg incorrect number of fields in control

pholdahl, 2020-09-08:
Hello good sir!

I have problems with getting the key bindings to work properly. Earlier Ive been using gapless, there I changed the keys.cfg to add numbers such as, 1 = pp-key-1 ...and so on. 

In the Control section of different shows in a media show I then e.x. added the command pp-key-1 play mp3, and it worked flawlessly, playing the track with the reference [mp3].

Now in pipresents beep, Im trying to do the same thing, but I get this error message: 

"Profile Error: incorrect number of fields in control pp-key-1 play mp3."

This does not happen if I change it e.x. pp-key-1 pause, or any other of the standard commands. So the binding of pp-key-1 seems to be good.

It just doesn't seem to like it when I add the reference of the track I want to play, but this worked very well in gapless.

Any ideas on why this occurs? Is it just a user error, or has anything changed in this section between gapless and beep?

KenT2, 2020-09-09:
If you are using a mediashow then play does not need a track reference because all it does is play the child track.

Play with track references are needed for radiobuttonshows and hyperlinkshows

pholdahl, 2020-09-09:
Hello!

I am so sorry, yes, this was only me, user error. Been a while since I´ve been working in this program, and forgot about the differences between radiobuttonshows and mediashows... 

Everything works as expected after changing to radiobuttonshow.

Thanks for the response!

closed 2020-09-09 by KenT2