KenT2/pipresents-beep #23: Radiobuttonshows that stay open in the background after being commanded to close.

pholdahl, 2020-09-12:
I am trying to make the same button, e.x. pp-key-1, with the same “play” command, play new video files, by redirecting the user from one radioshow to another, with media shows in between with “open” and “close” radioshow/media show as show control commands.…

I have a project like this, with no issues in Pipresents gapless...

In Beep, I have tried several combinations, and also the exact same build as I have in Pipresents gapless. But the same problem occurs.

The radioshows do not close, they stay open in the background. What happens then is that when I press the same button, e.x. pp-key-1, it starts the video on the current radioshow, and on the past (supposed to be closed) radioshow at the same time, and the videos are in conflict.

I have tried having show control “close” commands both on the radioshows themselves, on the tracks and on the mediashows, nothing seems to actually close the radioshows, except the "openexclusive" command.

The "openexclusive" command doesn’t really work for me, as I have a mediashow with music going on at the same time, with a static background image, to further hide the loading time when opening different shows and tracks.

The build with the controls and commands looks something like this:

Start
Start shows: mediashow1

mediashow1 with start, continue, single run, End trigger parameters: pp-key-1
Video.mov with show control at beginning: open bgmusic
Sound.mp3 with background image and show control command at beginning: open radiobuttonshow1

bgmusic
music.mp3 with background image.jpg

radiobuttonshow1 with First Track: [first], Controls: pp-key-1 play [next] and Show Control at End: close radiobuttonshow1
Video.mov [first]
Sound.mp3 [next] with background image and show control commands at beginning and end: open mediashow2, close radiobuttonshow1

mediashow2 with start, continue, single run, End trigger parameters: pp-key-1
Video.mov
Sound.mp3 with background image and show control command at beginning: open radiobuttonshow2

radiobuttonshow2 with open [first] track, controls pp-key-1 play [next] and Show Control at End close radiobuttonshow2
Video.mov [first]
Sound.mp3 [next] with background image and show control command at beginning and end: open mediashow3, close radiobuttonshow2

mediashow3 with start, continue, single run, End trigger parameters: pp-key-1
Video.mov
Sound.mp3 with background image and show control command at beginning: open radiobuttonshow3

radiobuttonshow3 with open [first] track, controls pp-key-1 play [next] and Show Control at End close radiobuttonshow3
Video.mov [first]
Sound.mp3 [next] with background image and show control command at beginning and end: open mediashow1, close radiobuttonshow3

In radiobuttonshow3 it goes back to the first mediashow, so the whole thing is a loop.

This build works without issues in pipresents gapless. Why does it not work the same way in beep?

KenT2, 2020-09-17:
Hi
I can see nothing obviously wrong with your profile and there is nothing I have intentionally modified in the show control area between gapless and beep so your profile should work in both.

Have you tried using the -d command line option. The ensuing debug output may reveal something not quite right.

Failing that could you develop a small profile that shows the problem, ideally with media from the examples, and send it to me either here or via the wordpress site contact page.

pholdahl, 2020-09-19:
Hello Ken!

I did what you asked with the example media, and the profile with its controls, worked as it should. So I did some more double checking between my new example profile, and the one that did not work as expected. And finally I found a show control command in the wrong place, making one of the radiobuttonshows reopen when it was not suppose to. I deleted this command, and everything is like it should be now.

Thanks so much for your response. The issue was clearly with me, and not pipresents...

closed 2020-09-19 by pholdahl