KenT2/pipresents-beep #24: start and stop a live show with one button

natanja, 2020-09-18:
Hi,

I have a project where one button should start a liveshow. when the liveshow is running the same button pressed should stop the liveshow. But that does not work. Only a start trigger works. But if I use it as end trigger also the start trigger does not work anymore. Some delay should do the trick? I don't understand why it overrules each other. It is a push button so only the falling command triggers.

KenT2, 2020-09-19:
Not sure exactly what your problem is but using the same symbolic name for start-trigger and end-trigger should work.
Using the same symbolic name for next-trigger and end-trigger, only end-trigger will work
Try running the profile with the -d command line option. The ensuing log might yield what is going wrong.

natanja, 2020-09-19:
I got it working!
I first had a radiobuttonshow with two liveshows in it. But then the end command didn't work with the same trigger.
With just one liveshow where the start and end input trigger is the same, it works like a charm!

Thanks for the tip about the -d. I will keep this in mind for a new challenge ;-)

Thank you!!

closed 2020-09-19 by natanja