KenT2/pipresents-beep #25: Profile Error:  xrand r display not recognised default

ShaoMingya, 2020-11-19:
Pi4B

cmd:
` python3 /home/pi/pipresents/pipresents.py -p pp_mediashow_1p4
`

error out:
```
Maps: {'DSI0': 0, 'HDMI0': 2, 'HDMI': 2, 'A/V': 3, 'HDMI1': 7} {0: 'DSI0', 2: 'HDMI0', 3: 'A/V', 7: 'HDMI1'}
Pi Model: 4
xrandr: Failed to get size of gamma for output default

Number of Displays - randr: 1
Displays Connected- randr: ['default']
Display Dimensions - randr: {'default': 1920} {'default': 1080}
Display Position - randr: {'default': 0} {'default': 0}
Display Rotation - randr: {'default': '0mm'}

Number of Displays - tvservice: 2
Displays Connected - tvservice: [2, 7]
Display Dimensions - tvservice: {2: 1920, 7: 1920} {2: 1080, 7: 1080}
1.44  Profile Error:  PiPresents:  xrand r display not recognised default
Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 889, in <module>
    pp = PiPresents()
  File "/home/pi/pipresents/pipresents.py", line 279, in __init__
    self.end('error',message)
  File "/home/pi/pipresents/pipresents.py", line 699, in end
    self.tidy_up()
  File "/home/pi/pipresents/pipresents.py", line 748, in tidy_up
    self.dm.terminate()
  File "/home/pi/pipresents/pp_displaymanager.py", line 224, in terminate
    self.terminate_backlight()
  File "/home/pi/pipresents/pp_displaymanager.py", line 1046, in terminate_backlight
    if self.backlight is not None:
AttributeError: 'DisplayManager' object has no attribute 'backlight'
```

KenT2, 2020-11-19:
Weird, I wonder what the 'default' display is.

what displays do you have connected?
Have you changed any configuration settings before running Pi Presents?

ShaoMingya, 2020-11-26:
> Weird, I wonder what the 'default' display is.
HDMI0.
> 
> what displays do you have connected?
Raspberry pie 4b, two HDMI ports each connected with a monitor;

> which operating system are you using?
Raspberry Pi OS with desktop and recommended software；

> Have you changed any configuration settings before running Pi Presents?
 I don't know where the configuration you are referring to.

KenT2, 2020-11-26:
Looks like you have a standard setup so I need to do a bit of debugging.  I did forget to ask which version of Raspberry Pi OS you are using. buster, stretch etc?

PP obtains its information about displays from the xrandr command. If you type this into  a terminal window you should see something like the output below. I would be interested in what you get. You can copy and paste the output using mouse right clicks and drag.

pi@SD1:~ $ xrandr
Screen 0: minimum 320 x 200, current 2944 x 1080, maximum 7680 x 7680
HDMI-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 509mm x 286mm
   1920x1080     60.00*+  50.00    59.94  
   1920x1080i    60.00    50.00    59.94  
   1680x1050     59.88  
   1280x1024     75.02    60.02  
   1440x900      59.90  
   1280x960      60.00  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    70.07    60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       75.00    72.81    66.67    60.00    59.94  
   720x400       70.08  
HDMI-2 connected 1024x768+1920+0 (normal left inverted right x axis y axis) 310mm x 230mm
   1024x768      75.03*+  70.07    60.00  
   1280x720      60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   640x480       75.00    72.81    66.67    59.94  
   720x400       70.08

ShaoMingya, 2020-11-27:
When "dtoverlay=vc4-fkms-v3d" is added to the "/boot/config.txt", pipresents can run correctly, but this makes "/dev /fb1" disappear. My own program needs to use "/dev /fb1", so it is contradictory and I don't know how to solve it.

And the version of Raspberry Pi OS is buster:
```
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
```

Here is the result of xrandr when "#dtoverlay=vc4-fkms-v3d":
```
pi@raspberrypi:~ $ xrandr 
xrandr: Failed to get size of gamma for output default
Screen 0: minimum 1920 x 1080, current 1920 x 1080, maximum 1920 x 1080
default connected 1920x1080+0+0 0mm x 0mm
   1920x1080      0.00* 
```

KenT2, 2020-11-27:
Your use of the legacy driver on a Pi4 is what is causing the problem as for a Pi4 PP expects the new driver which uses xrandr.

You might be able to fix the probelm my making PP think it using a Pi3.  DisplayManager.pi_model=3 in line 168 of pp_displaymanager.py will do this.  For a Pi3 PP uses only tvservice to find the monitors.

You might also need to modify framebuffer_priority and max_framebuffers in config.txt

closed 2021-03-07 by KenT2