KenT2/pipresents-beep #26: Control Shows by CLI

masseelch, 2021-02-15:
Hi Ken,

great, really great package. I got one question though: I try to create a little art show, where i need to be able to start / stop / select different shows from a remote machine. Is there a way to send commands to the running pipresents instance on the pi without using OSC?

KenT2, 2021-02-16:
OSC is probably the best. There is a utility pp_oscremote.py which has a show control gui, uses OSC, and will run on another Pi, probably on another Linux box. Possibly on Windows.

Other ways of remotely controlling a running instance of Pi Presents:
 - gpio - lots of short wires
 - the serial link I/O plugin - use a RS232 pair of wires or maybe USB and convertor.
 - the 'inputdevice' I/O plugin and a wireless remote control. Could be adapted for a IR remote.

The alternative is to have seperate profiles and run Pi Presents remotely using:
- The Pi Presents remote manager - pp_manager.py -  which is browser based
- Real VNC -
- remote SSH

masseelch, 2021-02-16:
Well, starting / stoping pipresents via remote SSH is what i wanted to achieve.

Is there a way to start pipresents via a SSH connection? So far i always get "PiPresents must be started from Desktop". Searching through the issues did not come up with a solution to circumvent this.

The problem I have with OSC is, that the place we set the art installation up in has very restrictive policy on their network. I am not able to use any other ports than 22 / 80 / 443.

masseelch, 2021-02-16:
Found out myself finally. For everyone landing here: 

```bash
DISPLAY=:0 DESKTOP_SESSION=LXDE-pi python3 /home/pi/pipresents/pipresents.py -p pp_artliveshow_1p4 -f -b
```
Runs the artliveshow example via a remote ssh connection on the latest buster.

closed 2021-02-16 by masseelch