KenT2/pipresents-beep #27: Artliveshow with cache - Symlinks

masseelch, 2021-02-17:
Hi again,

i am using the `artliveshow`-type and am having the following structure:

[image]

As you can see i have the folder `available_media` which contains all the media i can use for the show and am simply symlinking them in the `live_tracks_cache` (which are then getting copied to `live_tracks` after touching the  .lock-file). 

This prevents huge writes to the sd-card in case of frequent changes to the displayed media (since symlinks are smaller than a whole file) which will lead to longer lifetime.

It seems to work fine, at least it played through a night without errors and frequent changes to the media (done by a little script and a cron-job). 

Are there any reason which might advise me to not do it this way? If not, i think we should consider adding an example of this to the manual.

I like your package more and more :-)

KenT2, 2021-02-26:
I'm not a LInux expert but I imagine symlinks are just pointers to files so there should be no problems. I think its a good Idea and will add it to the manual.

KenT2, 2021-03-07:
In the latest manual

closed 2021-03-07 by KenT2