KenT2/pipresents-beep #28: Precise sync pulse output in artmediashow

whimsical-sam, 2021-02-26:
Hi,

I’m still experimenting but here’s my issue:

Goal: I would like to get the precise system timestamp at which a looping video began playing. 

- My original setup used python to (1) register a timestamp (2) send an OSC message to pipresents on a port which would (3) open a mediashow. The issue here of course ended up being that the time it took to load the file and actually start it was inconsistent, along with the fact that there is likely some amount for the osc message to be arrive, even using 127.0.01. This offset time did not seem consistent. 
- My second attempt switched to a looping artmediashow. Now, pp is set up as a master rather than a slave, and the media file is configured to send an osc command when it starts. The python script is setup to listen on the appropriate port (*nb: since the listening script is on the same machine, this did require changing the pp osc driver to listen for slaves on a hard coded port chosen to be different from the python script/slave port*). When the python script receives a message from pp that the file just started it registers a timestamp. The only latency here should be the osc message traveling within the pi itself, plus any gap there may be between the video starting and when the osc message is actually sent off by pp. I have not yet gotten to observe the accuracy of this since I only have vnc access to the machine until Monday, however I imagine there is still some degree of latency. I also haven’t tested yet if it is better to use the start or end command though they seemed more or less simultaneous to me. 

Most preferable to me though would be to just modify the existing pipresents distribution at the location where the new loop in the artmediashow actually begins, ie when the buffered second copy of the file begins. Just need to register the time stamp there, and then the time stamp itself can be sent off to wherever it needs to go (this can happen later if necessary since the timestamp was already registered). 

I was wondering if you could point me in the right direction as to the location in the code where that actually happens. 

Thanks so much, sorry for this super detailed issue!

whimsical-sam, 2021-02-27:
It looks like

KenT2, 2021-02-28:
The place to save the timetamp is pp_omxdriver_1p3.py Line 210 in def show() where the video is unpaused to show it.

You do not say how precise the timestamp has to be. PP pauses the video a few frames before the first, the few being a variable number.

whimsical-sam, 2021-02-28:
Thanks for the response! 

Just needs to be accurate within 100ms or better ideally. 

I’m syncing mobile devices to show subtitles for the film. Mobile devices synchronize their time to a web servers time using the ntp algorithm, pi uses the ntp algorithm to sync its time to the web server, pi sends the start time stamp to the server, and there you go.

whimsical-sam, 2021-02-28:
The OSC start command actually gets sent before the show() is called, however the timing difference there is minimal, so I ended up actually just taking the timestamp when the OSC command is executed and appending it to the osc message I am sending.  

Seems to be working well now.

closed 2021-02-28 by whimsical-sam