KenT2/pipresents-beep #29: hyperlinkshow with hyperlinkshow as subshow

linarkm, 2021-03-03:
Hi Ken, first of all thank you for developing this great tool! 
I am new to pipresents and I'm trying to create a hyperlinkshow which should work on a touchscreen. My question is now: is it possible to create a hyperlinkshow with a subshow that is also a hyperlinkshow? I would like to have a menu and a submenu which both work on the touchscreen? 
or is it not possible to have submenus in hyperlinkshows? 

I tried it but I always get an error message saying "first-track not found in medialist" 

I included a first and a home track in the sub-hyperlinkshow but it doesn't seem to work. 

Am I doing something wrong or is it just not possible to do this?

I use a raspi 4, pipresents beep and newest version of the raspian os with desktop, buster. 

Thanks and all the best,
Lina

KenT2, 2021-03-03:
Having a hyperlinkshow as a subshow of a hyperlinkshow works, I just tried it just to be sure.

As you say the sub show must have its own medialist with the home and first tracks. Maybe you have forgotten to give the tracks a track reference or they are missing in the show profile.

linarkm, 2021-03-04:
Hi Ken, thanks for the fast reply. Hmm included all the track references. But good to know that it generally works, I will go on looking for the error.

KenT2, 2021-03-04:
The error message should have the name of the track reference of the first track  it is looking for. This is blank in your original message which suggests you have not filled in the First Track field in the show.

closed 2021-07-04 by KenT2