KenT2/pipresents-beep #31: Profile Error: Touch driver is unable to find device raspberrypi-ts|

myxxmikeyxx, 2021-03-09:
Not sure what this error is. I assume it is because I am using the pi 7" Touch screen. I was getting the error before about the rpi-backlight so I did `sudo pip3 install rpi-backlight`. 
Now I am getting this error when trying to run then example code:
 `python3 /home/pi/pipresents/pipresents.py -p pp_mediashow_1p4 -f -b`
I also have tried the touchscreen example code: 
`python3 /home/pi/pipresents/pipresents.py -p pp_touchscreen_1p4 -f -b`

But I still get: `1.39  Profile Error:  PiPresents:  Touch driver is unable to find device raspberrypi-ts|`
😭 

Any help, I tried searching pypi.org for is it was another pip install but didn't find anything.

myxxmikeyxx, 2021-03-11:
[image]
Can I not use the updated one then? Or can I downgrade the driver? 
I was just trying to use PINN to install the older OS but I don't think it worked then. 

I did `cat /etc/os-release` plus I also did `lsb_release -a` 
I am on stretch not buster. 
https://pimylifeup.com/download-raspbian/

I can download the buster version version from [here ](https://downloads.raspberrypi.org/raspbian/images/)but that means my multi boot SD will not work :/
I am going to try to upgrade stretch to bust following [this guide](https://www.cyberciti.biz/faq/update-upgrade-debian-9-to-debian-10-buster/#:~:text=list%20using%20a%20text%20editor,running%3A%20sudo%20apt%20full%2Dupgrade)

myxxmikeyxx, 2021-03-11:
After updating it is working but I am getting a new error.
[image]
This is the new problem I am getting when running 
`python3 /home/pi/pipresents/pipresents.py -p pp_touchscreen_1p4 -f -b`
[image]
This is the other error I am getting when running
`python3 /home/pi/pipresents/pipresents.py -p pp_mediashow_1p4 -f -b`

I am seeing the raspberrpi-ts now when I ran `xinput --list`

KenT2, 2021-03-11:
The canvas error is because you do not have hdmi0 connected to a monitor. To correct this you need to change the Display field in the Show part of the profile.
The audio device error is for the same reason. It can be corrected in this version by selecting the A/V audio device for the audio track in Track Defaults tab of the show, and by selecting the A/V audio device in the task bar for VLC video tracks.
I am working on a new version which will allow the audio device to be selected from the profile in VlC Video tracks in addition to the audio tracks. I will also make these two error messages more meaningful.

myxxmikeyxx, 2021-03-15:
I am not sure what you mean when you say "selecting the A/V audio device for the audio track in Track Defaults tab of the show" .
I have opened the editor but I am not sure where you mean. Any screen shots or videos examples? 
I tried nano into the files but I didn't want to mess them up so I switched to the editor but this is my first time messing with it.

myxxmikeyxx, 2021-03-15:
I have found it. I didn't realize the setting was in the audio file. Well I know its not "in" the audio file, but you select the audio file and then select the output. 
Thank you for the help and sorry for the trouble.

closed 2021-03-15 by myxxmikeyxx