KenT2/pipresents-beep #33: pp_editor.desktop && pp_examples.desktop

myxxmikeyxx, 2021-03-15:
I have changed the .desktop files so they would work for me. 
Not sure if its because of my version or update but here they are.
I used `desktop-file-validate <file>` to make sure they are good. 
It gives a warning about having two categories that it could show up twice but I don't think that is a problem.

[Update PP Examples.txt](https://github.com/KenT2/pipresents-beep/files/6142942/Update.PP.Examples.txt)
[Updated PP Editor.txt](https://github.com/KenT2/pipresents-beep/files/6142943/Updated.PP.Editor.txt)

myxxmikeyxx, 2021-03-15:
I also used this as my reference.
https://linuxconfig.org/how-to-create-custom-desktop-files-for-launchers-on-linux
I just like to post what I used so others that might see this can see references and maybe learn or correct my errors.
😃

myxxmikeyxx, 2021-03-15:
Now the desktop presenter editor for some reason only works once. I tried doing just execute and execute in terminal but it still only works once then you have to reboot. My guess is that the python webserver is still running after closing chrome. Idk how to fix that.

myxxmikeyxx, 2021-03-15:
`pi@raspberrypi:~ $ /usr/bin/python3 /home/pi/pipresents/pp_web_editor.py -l
Found pp_web.cfg in  /home/pi/pipresents/pp_config/pp_web.cfg
Traceback (most recent call last):
  File "/home/pi/pipresents/pp_web_editor.py", line 1599, in <module>
    update_interval=0.3, start_browser=start_browser)
  File "/home/pi/pipresents/remi/server.py", line 900, in start
    s = Server(main_gui_class, start=True, **kwargs)
  File "/home/pi/pipresents/remi/server.py", line 791, in __init__
    self.start()
  File "/home/pi/pipresents/remi/server.py", line 809, in start
    self, self._certfile, self._keyfile, self._ssl_version, *self._userdata)
  File "/home/pi/pipresents/remi/server.py", line 737, in __init__
    HTTPServer.__init__(self, server_address, RequestHandlerClass)
  File "/usr/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.7/http/server.py", line 137, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
`

So it is still in use, might be able to add a pre-execution condition, I am currently looking into that.

myxxmikeyxx, 2021-03-15:
This is what I have done and it seems to work fine. I could not get two commands to run so I threw them into the script.
Updated PP Editor.desktop
Update PP Examples.desktop
pp_web_editor_kill.sh

chmod +x ./pipresents/pp_web_editor_kill.sh

[pp_web_editor_kill.txt](https://github.com/KenT2/pipresents-beep/files/6143679/pp_web_editor_kill.txt)
[Update PP Examples.txt](https://github.com/KenT2/pipresents-beep/files/6143680/Update.PP.Examples.txt)
[Updated PP Editor.txt](https://github.com/KenT2/pipresents-beep/files/6143681/Updated.PP.Editor.txt)

KenT2, 2021-03-17:
I think for security reasons the server hangs onto the address for a while unless you close the socket properly which your fix does.
I tend to start the editor from the terminal window and CTRL-C to terminate the server but only when I am developing editor code.

You do not need to close the server if you close the browser, just reopen the browser window to continue.

I'll have a look at the desktop files, they always worked for me except recently you get that execute and execute in terminal dialog. You can get rid of it by an option in the editor preferences.

myxxmikeyxx, 2021-03-17:
I found out you could but I just wanted to be able to use the desktop shortcut for launching and not have to reopen chrome if I close it and then go to history and localhost.
This was more of a suggestion change after I found a solution but I get not doing it because you are just closing the python webserver to reopen it.

closed 2021-03-30 by myxxmikeyxx