KenT2/pipresents-beep #35: make the start of unclutter configurable

masseelch, 2021-04-13:
I wrote a wrapper around PiPresents to start / stop / control PiPresents to manage the execution of different profiles. 
To reliably kill PiPresents I am using *nix feature to kill PiPresents by its ProcessGroup. In that way I won't have dangling / orphanged omxplayers around. 

My displays do start unclutter as a service, since i never need the cursor and the displays do other stuff than PiPresents too. But since PiPresents does start unclutter it is marked as child-process of PiPresents. So killing PiPresents in the way I do leads do unclutter being killed too. 

So i propose to add an option to toggle the start of unclutter when running PiPresents. 

I can draft a PR if you like.

KenT2, 2021-04-13:
I will put an option in the next issue which could be a while. 

Meanwhile just comment out pp_displaymanager.py line 566 - os.system('unclutter > /dev/null 2>&1 &')

KenT2, 2021-04-13:
sending SIGTERM to PP should shut it down successfullly. Hopefully with no dangling omxplayers. But I guess it will terminate unclutter even though it is started with an &

masseelch, 2021-04-14:
Sending SIGTERM was what I did first. That sometimes is leaving omxplayer running in the background. If an error of some kind occurs and PiPresents shows a pop-up regarding that error SIGTERM will not work either. That is why I am doing it with pgids.

Commenting out the line works like a charm ☺️

KenT2, 2021-06-26:
In 1.4.5e

closed 2021-06-26 by KenT2

masseelch, 2021-06-26:
Great! I will update ASAP! Thank you very very much!!