KenT2/pipresents-beep #36: Question: Show Control on End and OSC Commands

masseelch, 2021-04-15:
Hey again :grin: 

I have a profile configured with 3 shows: `loop`,`spear` and `bow`. `loop` is meant to be played on loop. If the user presses the Spear-Button an OSC command is sent to `openexclusive` the show `spear. A Bow-Button opens show `bow` the same way. 

The shows `spear` and `bow` are configured to run once, then they have a **Show Control at End** command set to `open loop`. 

This works like a charm until the users presses the **SPEAR** button while the `bow` show is running. This triggers the `openexclusive` OSC command which in turn triggers the **Show Control at End** command and the `loop` show is played. But I want the `spear` show to play. How can I achieve that? 

I tried opening `spear` and `bow` without closing `loop` but the shows do not alway take the whole screen and one can see parts of `loop` running in the background.

masseelch, 2021-04-15:
> I tried opening spear and bow without closing loop but the shows do not alway take the whole screen and one can see parts of loop running in the background.

My second try was to give the `bow` and `spear` shows a black background to cover up the `loop` show in the background. This, however, is not working too, or I am just misconfiguring it.

KenT2, 2021-04-15:
I think you might be overcomplicating things.
The radiobuttonshow was designed to do what you want.
Make the 'First' track of the readiobuttonshow a mediashow and spear and bow shows that are played by the buttons.

A radibuttonshow is special in that input events are not passed down to subshows so buttons for spear and bow and stop are available while in the subshows (including the looping initial show). The downside of this is that all of the subshows cannot have controls.

This special behaviour can be turned of by Controls in Subshows.

masseelch, 2021-04-16:
Ah, I get the idea. But I cannot seem to understand how I can send the start / stop / return commands via osc :thinking:

masseelch, 2021-04-16:
NVM, figures it out. I need to send an input event to the radiobuttonshow and have that declared in the controls field. Thank you very much for this awesome piece of software and your advise. 

I will try to get a donation out of our dean for you :smile:

closed 2021-04-16 by masseelch