KenT2/pipresents-beep #38: VLC hangs a lot

masseelch, 2021-04-20:
Hey there. I just switched from omx-player to vlc and am getting a lot of freezes. Roughly every other rotation of the artliveshow vlc just hangs somewhere in the video and shows one frame forever. 

If you need any more information I am happy to provide them. For now I will keep using the last version of PiPresents still using omx-player.

KenT2, 2021-04-20:
I have had another report of similar problems with artliveshow. It works fine here with the videos used by the examples.

Could you play the video using the command PP sends to vlc:

cvlc --quiet --no-xlib --vout mmal_vout --mmal-vout-transparent --play-and-pause  --mmal-vout-window 400x400+0+0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu  xthresh.mp4

Also could you try the liveshow instead of artliveshow. The only difference will be a paused video at the end of each track

If you want to go back to omxplayer then edit line 301 of pp_livelist.py from 'vlc' to 'video'

masseelch, 2021-04-21:
I ran the command like 15 times, everything works. I play the video in fullscreen though (option: display). I will try the liveshow instead.

masseelch, 2021-04-21:
The same on liveshow. But it is not, as I first thought, in the middle of a video. The show sometimes does not restart when it reached its end. This seems to hapen only if the last track is a video. If i have an image like zzz.png (which then is the last entry) the restart seems to reliably happen.

masseelch, 2021-04-21:
I get this with omxplayer. To me it looks like a bug of some sort keeps the show playing while PP thinks it should be paused?

log="233.18  Warning:  OMXDriver_3019810608:  Failed to test paused - dbus exception: Message recipient disconnected from message bus without replying"
log="pause at end failed, probably because of delay after detection, just run on"
log="unpause for show False"
log="unpause for show False"
log="348.63  Warning:  OMXDriver_3019810128:  Failed to test paused - dbus exception: Message recipient disconnected from message bus without replying"
log="pause at end failed, probably because of delay after detection, just run on"
log="unpause for show False"

KenT2, 2021-04-21:
PP pauses a video before the end of a track, even in an artshow. The length of time before the end is a magic number as omxplayer does not have proper handshaking. This time might be too small and PP is trying to do the pause after the track has finished.
To increase the time edit line 122 of pp_omxdriver.py  to increase the number from 350000
The number is microseconds and a frame at 25 frames /sec happens every 40000 microsecods so try going up in these increments.

What Pi are you using?
It might be worth passing your video through something like Handshake. People have had problems with videos over the year and often they are homemade ones.

masseelch, 2021-04-21:
This worked yes. The video is BigBuckBunny downloaded from the official site. 

The problem with vlc still remains even with liveshow. I tried with handshaked videos but even then I get a maximum of 3 repeats until the show stops and does not repeat. PP hangs completely by the way. I can neither start the tracks with my attached buttons not do any show control commands work. 

There are no errors in the logs.

KenT2, 2021-04-21:
Which version of BBB are you using - dimensions, fps and file extesnsion - link if possible. BBB.org seems to be broken at the moment when I tried to download the 1080p 30fps .mp4 version

I have an old 720 version which I am trying at the moment.

masseelch, 2021-04-22:
[This](https://file.io/S3ELOAWmXtc2) is the file i am using.

masseelch, 2021-04-22:
> To increase the time edit line 122 of pp_omxdriver.py to increase the number from 350000
> The number is microseconds and a frame at 25 frames /sec happens every 40000 microsecods so try going up in these increments.

It started working on 430000, if you wanna know.

KenT2, 2021-04-23:
Thanks for the videos. I have run BBB overnight with VLC, and with omxplayer for a few hours with no problems. 
Just added Jellyfish
I am beginning to suspect a difference in our setups
Mine:
Pi4, RPi OS Buster, GPU_MEM = 252

masseelch, 2021-05-05:
I try with your setup and will report

masseelch, 2021-05-10:
If i only add Jellyfish there are no problemns either, Its when i have a mix of videos and images, when the hang occurs.

sky-love, 2021-05-15:
I had the same problem with VLC hanging up all the time, especially during longer video loops (10min). Shorter clips seemed to be working fine. I switched to OMX player, configuring my show right in the Editor, so didn't even modify any code, and now all my clips are looping flawlessly and the performance seems better too, not so jerky between frames anymore. On this note I am curious why the OMX player got deprecated and what does this mean for the future and any possible upgrades/updates? Thanks for a great program!

KenT2, 2021-05-16:
@sky-love 

The reason omxplayer is being deprecated is stated here. 

https://github.com/popcornmix/omxplayer  

While omxplayer still works with PP with a few limitaions there will come a time when Pi Towers modifies RPi OS so omxplayer stops working completely without notice. I therefore added VLC to PP expecting there would be a few latent bugs so I kept omxplayer in.
The only time you need to modify the code to revert to omxplayer is for a liveshow or artliveshow. For all other types of show the editor can be used to select vlc or video (omxplayer) tracks although out of the box everything selects VLC.

VLC hanging all the time in the middle of a video would indicate that the problem is with VLC as the only time that PP interacts with a video is at the beginning or end or, in the case of an art..show, in the first second or so when it loads the next track ready for playing. PP hanging at the end of a video is much more likely to be caused by PP. (missing the signal for the end of the track.)

Please check that  GPU-MEM is set to 255 or thereabouts
It would be interesting to run your videos using VLC from the desktop in fullscreen mode (so that it uses MMAL), or as PP would run them from a terminal:

cvlc --quiet --no-xlib --vout mmal_vout --mmal-vout-transparent --play-and-pause --mmal-vout-window 400x400+0+0 --mmal-layer=1 --mmal-display=HDMI-1 --mmal-vout-transform=0 --no-spu xthresh.mp4

Change 400*400 to your screen size for full screen, with care as recovery is not straight forward.

 If you have problems which are independent of PP then report them here [https://github.com/RPi-Distro/vlc/issues](url)

My problem is to recreate people's problems with my setup. I've failed so far :-(

KenT2, 2021-05-25:
Lots of experimentation in the last couple of days. For some reason GPUMEM got set to 152 on my rig and this caused the problems that people have been reporting to appear :-)

One thing is certain -  that VLC uses GPU memory and you must set this to at least 256 for PP to work reliably with 1080p videos and VLC.

I have been testing with the Jellyfish and Big Buck Bunny videos that you supplied. I think the problem is that the videos are 1920*1080 (1080p) and the Raspberry Pi is not good at playing them. This is made worse by Pi Presents because at times it plays two videos at the same time. 

I used Handbrake to convert the above two videos to 720p. Using these videos everything worked fine, as it has done with all the example videos, none of which are 1080p

If you are having problems with 1080p videos:

a. Try setting GPUMEM to 512.

b. In a medisahow/liveshow the problem occurs if 2 1080p videos are next to each other. This is because gapless operation requires 2 copies of VLC to be running at the same time, one paused at the end of a track while the other is loading the next track. So either seperate the videos with anything other than a 1080p video or set 'Freeze at End' to 'No' in the first of the abutting tracks. There will be a gap between the tracks but 1080p works. The manifestation of the problem is that the second of the abutting videos plays but its display remains black so it looks like it is not displaying

d. In artliveshow/artmediashows the second video is loaded during the playing of the first video so seperating the two 1080p videos is the only solution.

c. Don't show two 1080p videos at the same time. This is probably something you would not want to do but it is something I test. If you do the effect is hdmi0 intermittently  going blank. I have demostrated this without PP, just running the CVLC command.

I have not seen any pauses or stutters.

I need to experiment with omxplayer and 1080p videos but it sounds like it is working OK.

KenT2, 2021-05-29:
Have now run the 1080p BBB and Jellyfish videos you supplied back to back in a mediashow, liveshow and  artliveshow with no problems. With  a  GPUMEM of 200 things are fine; lower than that things go wrong.

masseelch, 2021-06-01:
I can confirm above 200 there are less problems. For me a value above 300 did work flawlessly. Since I have the 8GB Version, I can set the GPU to about 1024MB without any problems. I am thankful for your work and bow to your wisdom!

masseelch, 2021-06-03:
Well, I was too early. Tonight the vlc did hang again. GPU at 512 MB. Is there a way to have PP log vlc? (Starting without --quiet)

KenT2, 2021-06-10:
I cannot see an easy way with PP to remove --quiet. However I have attached a small file which will run the command that PP uses and play a video in a loop. If you can your videso to fail with this it would eliminate PP.

[disp.sh.txt](https://github.com/KenT2/pipresents-beep/files/6630753/disp.sh.txt)

I have also done a couple of upissues of PP which fix a couple of bugs in VLC player that might improve things.

There have been reports on the forums that people are having similar problems as you but with 4k videos so you are not alone.

KenT2, 2021-07-05:
Possible solution
At the end of /boot/config.txt add
core_freq_min=500

See: [https://github.com/RPi-Distro/vlc/issues/46](url)

masseelch, 2021-07-05:
> I cannot see an easy way with PP to remove --quiet. However I have attached a small file which will run the command that PP uses and play a video in a loop. If you can your videso to fail with this [...]

This does fail my videos. So PP is not the issue here.

masseelch, 2021-07-05:
> Possible solution
> At the end of /boot/config.txt add
> core_freq_min=500
> 
> See: [https://github.com/RPi-Distro/vlc/issues/46](url)

I moved to omxplayer for now.  

FYI, if you are interessted: The project is shipped to dean and will be displayed from 12th July until 18th July and only after that I know if it will be displayed again or not. Its an art display for a master thesis. If this will be presented for a wider audience, which might led to some form of income, I will let you know. There is something in the licence about charging a royal in case of profits.

closed 2023-02-25 by KenT2