KenT2/pipresents-beep #39: Question: List Empty Track for Artliveshow

masseelch, 2021-04-20:
Me again, sorry to bother you so much lately :-)

There is an option to show a default track for a liveshow (List Empty Track). But this option does not seem to work on artliveshow. Is this just a bug or is this feature not implemented? If not, how can I be of help enabling this?

KenT2, 2021-04-20:
The feature is not implemented. 

Do try the liveshow. The only difference is a short freeze at the end of each track which could be disguised by a suitable end to the video.

masseelch, 2021-04-20:
You mean like fading to black at end / fading from black at start? The freeze would happen at the last black frame and it looks like a fade, yes?

KenT2, 2021-04-21:
yes, or just a stationary scene at the end. Fade to black would disguise the very tiny gap where the background is shown

masseelch, 2021-05-07:
Works like a charm. Thank you :-) 

BTW, do you have a Buy Me A Coffee Link or sth?

closed 2021-05-07 by masseelch