KenT2/pipresents-beep #44: help reading Arduino

fullmakeralchemist, 2021-07-03:
Hi I tried to use your example [ppserialcontrol](https://github.com/KenT2/pipresents-beep-examples/blob/master/pp_home/pp_profiles/pp_serialcontrol_1p4/pp_io_config/pp_serialdriver.cfg)

Basically I put everything ok with the Serial Port, but it doesn't run.
I will show you in python what I did in python with another code:

```
ser = serial.Serial()
        #ser.port = "/dev/ttyUSB0"
        ser.port = "/dev/ttyACM0"
        #ser.port = "/dev/ttyS2"
        ser.baudrate = 9600
        ser.bytesize = serial.EIGHTBITS #number of bits per bytes
        ser.parity = serial.PARITY_NONE #set parity check: no parity
        ser.stopbits = serial.STOPBITS_ONE #number of stop bits
        #ser.timeout = None          #block read
        ser.timeout = 0.01            #non-block read
        #ser.timeout = 2              #timeout block read
        ser.xonxoff = False     #disable software flow control
        ser.rtscts = False     #disable hardware (RTS/CTS) flow control
        ser.dsrdtr = False       #disable hardware (DSR/DTR) flow control
        ser.writeTimeout = 2     #timeout for write
    
        ##    Print sprite_list dans la console
        #    for sprite_line in sprite_list:
        #        print(sprite_line)
    
        ser.open()

       response = ser.read();

        if (response==b'1'):
            print("playvideo1") #first condition I need
        elif(response==b'2'):
            print("playvideo2") #second condition that I need 
        elif(response==b'3'):
            print("playvideo3") #third condition that I need

```

There is some way you can tell me in what I'm wrong? 
or maybe I just need to play the video with pure python code?

I use the same port values on this code that in the pp_serialdriver.cfg

KenT2, 2021-07-04:
The only difference I can see is that PP sets ser.timeout = 0 and pols for input every 50mS.
If you add the -d option to python3 pipresents.py there will be a log which says the serial driver is active and will indicate if an input event is received when a character is sent.

Otherwise put some print statements in /pipresents/pp_io_plugins/pp_serialdriiver.py around line 174

fullmakeralchemist, 2021-07-04:
> The only difference I can see is that PP sets ser.timeout = 0 and pols for input every 50mS.
> If you add the -d option to python3 pipresents.py there will be a log which says the serial driver is [...]

Thanks I will try it.

fullmakeralchemist, 2021-07-04:
It gave me this error trying to run this ine  

python3 /home/pi/pipresents/pipresents.py -p pp_serialdisplay_1p4

erroe

File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port ttyACM0: [Errno 2] No such file or directory: 'ttyACM0'

KenT2, 2021-07-04:
I would expect the error to be /dev/ttyACM0
Your connected USB serial should appear in the directory /dev. There are lots of hits on Google about problems with it.

closed 2023-02-25 by KenT2