KenT2/pipresents-beep #45: vlc-image-duration ignored?

ToddRK, 2021-07-31:
I downloaded the latest version of Pipresents 1.4.6 and the latest python-vlc and it seems that every video I play in a mediashow using vlc is limited to 20 seconds regardless of the vlc-image-duration setting.  I tried many values from 0 to 10000 but all videos still only play for 20 seconds.  I don't see the same issue on other Raspberry Pis using Pipresents 1.4.5 and possibly older versions of python-vlc.

I tried dumping the value of self.vlc_image_duration_text in pp_vlcplayer.py and it seems to be set properly so I'm thinking it might be the latest python-vlc causing the issue but I'm not certain of this.

KenT2, 2021-07-31:
vlc image duration des not affect videos. It is only for still images (.jpg etc) which VLc can play. Sounds like you have set track timeout in the mediashow profile.
Does pp_mediashow_1p4 example work OK ?

ToddRK, 2021-07-31:
I just tried pp_mediashow_1p4 and it works OK.  I went back and checked all the settings in the profile and found that I'd somehow set the repeat interval to 25 seconds so setting it to 0 fixed the problem.

Thanks for your help with this.

I have to mention that I'm setting up many videos triggered by GPIO inputs for my local museum and I'm surprised it runs fine from a RPi Zero even though some of the videos are 1.5GB.  You've done an excellent job making the program versatile but still able to run on such simple hardware.  Thanks for all your efforts!

closed 2021-07-31 by ToddRK