KenT2/pipresents-beep #48: Videos and Images in different layers

AlainCx, 2022-02-03:
Hello,
I'm starting a new project involving layers and I'm not familiar with this in Pipresents.
I want to display small images above playing videos. The images will be triggered touching some areas on a touchscreen while a video is playing, then closed touching them (or after a few seconds) without interrupting the video.
Is it possible with pipresents ?
With a Radiobuttonshow ?
With VLC or OMX player ?
Thanks for your help !

KenT2, 2022-02-03:
Unfortunately you cannot display images above videos. It is a limitation or the Raspberry Pi not of Pi Presents. No difference between VLC and omxplayer in this respect. 

You could turn images into a video offline and display on a different layer. You can display images elsewhere on the screen or on a second display triggered by a touch area that is in the area of the video. 

A Radiobuttonshow would not work because the video would be stopped. I think the best way is to use a mediashow for the video and  'Show Control on Event' to start a new show.

Actually you have given me an idea for a new type of show which you would do similar to this in a more streamlined fashion.

AlainCx, 2022-02-03:
Thank you very much for you speedy answer !

So it could work if I turn my images into looped videos inside a subshow in a Mediashow right ?
The looped videos could be stopped and disappear with a touch ?

If so I think I could build my project with pipresents !

KenT2, 2022-02-03:
You can define many 'click areas' each playing a different video.

You are on the right track but you cannot use a subshow of a mediashow (a child track) as only one of these is allowed per show. What you need to do is to open a new top level show using 'show control on event'.

Not sure about stopping a looped video by touch either as you cannot associate more than one command with a click area. 
The other possibility is the stop command is associated with the click area in the video track but I don't think this will be reliable   as an input event is passed to all shows.

Best to use a single run track or show a stop button when the track is playing.

KenT2, 2022-02-03:
I was interested to see if what you wanted could be done so I knocked up a profile which is attached.
To get the videos to stop was easy, just define a click area that is used only in the appropriate video show.
I got caught out because I did not include pp-spawnx no-command entries in the controls fields to enable the click area. Its not clear in the manual  that you need to do this.
[stream.zip](https://github.com/KenT2/pipresents-beep/files/7995920/stream.zip)
.

KenT2, 2022-02-03:
No problems. Doing this type of thing is a new idea for me.

Forgot to say the click areas are top left and top right of the video
You need to run fullscreen to get the registration correct
You can hide the click area flashes by making the areas  transparent