KenT2/pipresents-beep #50: 4K mediashow hang at the end f the video

yadnet, 2022-04-09:
Hi,
I'm using the lastest version of pipresent with latest recommended version of Pi OS. (Following install guide)

I have a setup with a mediashow that only play a 4K video at the moment and loop.
The show is starting well and play video smoothly on a Pi4 4GB using HDMI0.
At the end of the video, (aprox 5s. before the end) the screen goes black and the TV loose the signal and finaly goes in standby.
By looking at the tasks, a new instance of pp_vlcdriver.py is well started but it does not output anything.

I have a video memory set to 512 and core_freq_min=500.

I think there is something that goes wrong when trying to launch two instance of vlc for the transition, but I don't know how to overcome that. It's not a problem for me if there is a black gap between the loop of the video.

Thanks for your advice.

Best regards,
Yannick

KenT2, 2022-04-10:
I suspect the problem is that the Pi is struggling with two 4K videos. You can stop the overlap by setting 'freeze at end' to No in the video track.

yadnet, 2022-04-10:
I've tryed that, but seems that the result is the same.
Atteched the files of the profile. Can send the video as well if you like
[mediashow.json.txt](https://github.com/KenT2/pipresents-beep/files/8458760/mediashow.json.txt)
[pp_showlist.json.txt](https://github.com/KenT2/pipresents-beep/files/8458761/pp_showlist.json.txt)

KenT2, 2022-04-12:
Your profile is good. I assume you are using RI OS Legacy.

If the problem occurs 5 seconds before the end the  it is unlikely to be running 2 videos anyway as the overlap should start less than a second before the end.

I had problems with 1080p videos which is here
https://github.com/RPi-Distro/vlc/issues/46
but it looks like you might have seen that.

Does the video play OK using VLC from the desktop when fullscreen?

It might be worth Googling. I found this https://www.makeuseof.com/run-your-raspberry-pi-at-4k-60hz/

yadnet, 2022-04-16:
Yes, I'm using OS Legacy as stated in the doc.

I've done some new tests this moning in the museum to find more info (as a volunteer in not always there).
To confirm if vlc is correctly running, I've tested to start cvlc --loop --fullscreen and it's working fine on my three screens.

I've modified the show to add a full screen image during 3 seconds after the video before loop.
Result is that sometimes I have the image and then the show restart correctly, sometimes at the end of the video the screen is back and even don't switch to display the image.
One strange thing is that the TV seems to disconnect right after the end of the video as it display the 'no signal' warning. I don't know how hdmi link is managed inside your process.

One experiment done with success was:
- Play the show until the end of the video
- confirm that the TV lost signal
- check running tasks and confirm that the pp_vlcplayer is running. I can confirm that vlc seems to work ok, but no HDMI output...
- kill pp_vlcplayer
some time after the video restart ok. (I think that pipresent detect this condition and relaunch the pp_vlcplayer...)

So we have maybe somethink linked with HDMI output selection. Note that I'm on PI4 and used the HDMI0

Hope those info can help.

KenT2, 2022-04-16:
Yes your report was very useful its given me some good ideas.

I do not have a 4k monitor or video so I cannot debug the problem myself. However I have a good idea what the problem might be.
pp_vlcdriver.py runs as a separate process. It outputs to the display using vout=mmal_vout not x windows and to get rid of the extraneous desktop display it is run with display=  to hide the x window desktop.

I think maybe PP is not detecting the end of the video from VLC and is hence not moving on. If it is displaying full screen then nothing will then display and the monitor will display no signal. You could verify this by displaying the video in a window. You should then see PP's background and the video will display a black rectangle.

If this is the case, why is PP not detecting the end of the track? It is VLC that does this detection. I need you to turn on logging of pp_vlcdriver.py to determine whether this is the case.

In pp_vlcdriver.py line 377 change enabled=False to enabled = True. There will then be a log in pipresents/pp_logs/vlcdriver_log.txt

With freeze at end = 'no' line 261 should output 'ended with no pause' if end detection is working.

It would be interesting to see what happens if freeze_at_end = yes. If you uncomment the logging statement in line 250 we can see why that did not work for you.

yadnet, 2022-04-17:
Hi Ken,

I've been able to make some tests this morning. Here is the log of the first two rounds (duration of the video is 10"04):

Cannot play the video in a windows as 4K seems only supported fullscreen (even with vlc GUI alone).
So only able to make fullscreen tests:

The test with the freeze at end : NO:
`6.684788  191719     init VLC instance
6.736540  191719     iopts:  --mmal-vout-window 3840x2160+0+0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu -
-aout=pulse --image-duration 5
6.786633  191719     pauseopts:  no no
6.836736  191719     track:  /home/pi/pp_home/pp_profiles/murchrono/Video-Ecran.mp4
6.886835  191719     Command:  load
6.886894  191719     Instance Options:   --quiet --no-xlib --vout mmal_vout --mmal-vout-transparent  --mmal-vout-window 3840x2160+0+0
--mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu --aout=pulse --image-duration 5
7.107810  191719     track length 604139
7.113847  191719     Command:  get-size
7.114213  191719     Size: 3840 2160
7.166265  191719     Command:  play
7.240153  191719     track frozen at start at:  0 -1
7.240287  191719     load-ok at:  0
7.347916  191719     device:  alsa_output.platform-bcm2835_audio.digital-stereo
7.420400  191719     vol:  0
7.453210  191719     Command:  show
7.453430  191719     freeze off, start showing
7.454050  191719     show loop start
7.503927  191719     vol:  0
624.114073  191719     ended with no pause
633.337938  192345     init VLC instance
633.388793  192345     iopts:  --mmal-vout-window 3840x2160+0+0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu
 --aout=pulse --image-duration 5
633.438869  192345     pauseopts:  no no
633.488970  192345     track:  /home/pi/pp_home/pp_profiles/murchrono/Video-Ecran.mp4
633.539051  192345     Command:  load
633.539092  192345     Instance Options:   --quiet --no-xlib --vout mmal_vout --mmal-vout-transparent  --mmal-vout-window 3840x2160+0+
0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu --aout=pulse --image-duration 5
633.607728  192345     track length 604139
633.614406  192345     Command:  get-size
633.614753  192345     Size: 3840 2160
633.665527  192345     Command:  play
633.700861  192345     track frozen at start at:  0 -1
633.700943  192345     load-ok at:  0
633.768677  192345     device:  alsa_output.platform-bcm2835_audio.digital-stereo
633.818580  192345     vol:  0
634.015047  192345     Command:  show
634.015170  192345     freeze off, start showing
634.015550  192345     show loop start
634.065361  192345     vol:  0`

The test with the freeze at end : YES:
`6.638291  190530     init VLC instance
6.689813  190530     iopts:  --mmal-vout-window 3840x2160+0+0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu -
-aout=pulse --image-duration 5
6.739931  190530     pauseopts:  no yes
6.790031  190530     track:  /home/pi/pp_home/pp_profiles/murchrono/Video-Ecran.mp4
6.840129  190530     Command:  load
6.840179  190530     Instance Options:   --quiet --no-xlib --vout mmal_vout --mmal-vout-transparent --play-and-pause  --mmal-vout-wind
ow 3840x2160+0+0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu --aout=pulse --image-duration 5
7.062397  190530     track length 604139
7.068433  190530     Command:  get-size
7.070178  190530     Size: 3840 2160
7.121158  190530     Command:  play
7.189325  190530     track frozen at start at:  0 -1
7.189408  190530     load-ok at:  0
7.301177  190530     device:  alsa_output.platform-bcm2835_audio.digital-stereo
7.351533  190530     vol:  0
7.402879  190530     Command:  show
7.403082  190530     freeze off, start showing
7.403602  190530     show loop start
7.454040  190530     vol:  0
624.108510  190530     paused at end at
627.606833  190530     Command:  close
633.105427  191157     init VLC instance
633.156271  191157     iopts:  --mmal-vout-window 3840x2160+0+0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu
 --aout=pulse --image-duration 5
633.206352  191157     pauseopts:  no yes
633.256448  191157     track:  /home/pi/pp_home/pp_profiles/murchrono/Video-Ecran.mp4
633.306532  191157     Command:  load
633.306572  191157     Instance Options:   --quiet --no-xlib --vout mmal_vout --mmal-vout-transparent --play-and-pause  --mmal-vout-wi
ndow 3840x2160+0+0 --mmal-layer=1 --mmal-display=HDMI-1  --mmal-vout-transform=0 --no-spu --aout=pulse --image-duration 5
633.374965  191157     track length 604139
633.379021  191157     Command:  get-size
633.379290  191157     Size: 3840 2160
633.430059  191157     Command:  play
633.465287  191157     track frozen at start at:  0 -1
633.465374  191157     load-ok at:  0
633.533121  191157     device:  alsa_output.platform-bcm2835_audio.digital-stereo
633.584254  191157     vol:  0
633.786051  191157     Command:  show
633.786176  191157     freeze off, start showing
633.788101  191157     show loop start
633.836426  191157     vol:  0
`

Both tests was done with the following mediashow config with the image at the end.

[mediashow.json (2).txt](https://github.com/KenT2/pipresents-beep/files/8501406/mediashow.json.2.txt)

Note that I've tested another scenario with a reduced video size of only 1"35 lenght video and it seems to be able to loop many times before having the black screen. I don't really understand why... ?  

Best regards,
Yannick

KenT2, 2022-04-17:
Thanks for the logs.
As I suspected in both cases the first time playing the video VLC reports it has ended. The second time it does not so PP hangs waiting for it. Sounds like VLC is intermittently broken when playing 4K. 

Have you tried the 4K mod to config.txt I suggested last time   (url) https://www.makeuseof.com/run-your-raspberry-pi-at-4k-60hz (/url)

Can you use a program to convert the 4k video to full HD?

yadnet, 2022-05-01:
Hi Ken,
Sorry I failed to get access to the screens to make some tests due to a lack of time to do so.
I will make tests next weekend.

closed 2023-02-25 by KenT2