KenT2/pipresents-beep #51: Full screen

wwagonman, 2022-04-25:
I have been using pipresents for a number of years. Finally decided to start a fresh install with Buster/Beep. 

Everything installed fine, set up a new profile (mediashow) and it plays fine. My problem is that the screen blanks and the video plays, but the video is in a small format in the upper left of the screen. 

The example profiles work fine full screen  so it is clear that it is a setup problem. I have poured though the example profiles and the manual and am getting nowhere. 

Thanks!

Lee

KenT2, 2022-04-26:
The VLC Video Track profile has a field called VLC Window. Set this to display. See manual for other options
Hope this helps

wwagonman, 2022-04-26:
Thanks Ken, you got me on the right track. All good now. 

Lee

closed 2022-04-27 by KenT2