KenT2/pipresents-beep #52: Provide CR at the end of serialdriver output

yadnet, 2022-05-01:
Hi,
I'm trying to output <CR>at the end of serialdriver output string without success.
Is seems that we can't do that. Even if I'm trying to add \n at the end of the command config, python read it as \\n.

The only workaround found is to append a byte before sending the string in pp_serialdriver.py

```
bytes.append(0x0D)
bytes_sent = pp_serialdriver.ser.write(bytes)
```

Do you know another solution ?

Best regards,
Yannick

KenT2, 2022-05-02:
Yes the problem is in the string to byte conversion in lines 309 onward. The solution is probably in the link below or similar links which properly handles escape characters.

https://stackoverflow.com/questions/4020539/process-escape-sequences-in-a-string-in-python

I need to get my head around all this encoding and escaping  stuff. If you fancy doing it that would be great. Meanwhile your solution seems good, or use the bytes alternative method in the driver.

KenT2, 2022-06-30:
fixed in version 1.4.6d. Escaped characters such as \n \r now send the correct bytes. Not tested with an actual serial link.

closed 2022-06-30 by KenT2