KenT2/pipresents-beep #54: touchscreen : touch areas issue

AlainCx, 2022-06-22:
Hello,
I'm finishing a project combining several radiobuttonshows controled by a touchscreen. 
The menus are designed in Photohop and several touch areas are set up in "screen.cfg"
The touch areas don't overlap as I thought it might be a problem (see below)
[image]

I found a strange issue : in each subshow touch areas n° 5, 6 and 7 don't work except if I remove the 3 touch areas corresponding to the 3 choices in the top menu from the "screen.cfg" file. I made a test replacing touch controls in the top menu with key 1, 2 and 3. In this case all the 9 touch areas of the subshows are working.

Here is the project if it can help understanding the problem.
[IMPACT_BORNE.zip](https://github.com/KenT2/pipresents-beep/files/8959083/IMPACT_BORNE.zip)

Thanks for any help

Alain

AlainCx, 2022-06-24:
Thanks Ken.
I used the contact form via pipresents.wordpress.com earlier.

KenT2, 2022-06-30:
Fixed in version 1.4.6d

closed 2022-06-30 by KenT2