KenT2/pipresents-beep #55: Show newest file first in a liveshow

dominiquegrota, 2022-06-29:
Hi, 
I am very glad to use your pipresents ! it is really a good product. My usecase is : I have a pibooth to take pictures, and I rsync my pictures after each new selfy to a raspberry with your pipresents. 
I can slideshow all pictures automatically, but the new picture will appear at the end of the slideshow. I would like to show the newest file first. I only find the shuffle or ordered option. Is there a way to use somthing like "ordered desc" or "ordered -1"
Thank's for your answer and thanks for the job !
Dominique

KenT2, 2022-06-30:
You are in luck. I was just about to make a new version of Beep so I slipped this in to the version I uploaded today. The option is reverse.

Tell me if it does not work

dominiquegrota, 2022-08-28:
Hi ! 
Sorry for my very late answer..... I do recieve the notification, and then I went on hollidays :-/

It is excactly what I wanted !!! Thank's So mutch :-D 
I updated pipresents, and it is working well !

Thank's again. 
You can close the issue.

Dominque

closed 2022-08-28 by KenT2