KenT2/pipresents-beep #56: Cookie behaviour in Chrome Web Track

PublicBenemy, 2022-08-30:
Hello and thank you for this wonderful piece of software. Is it possible that the chrome web track does not store cookies? If so, is that intentional? When I run a show and do configurations on a site that should remain, then when I stop the profile, they're vanished the next time I run that profile again.

I am using the hardware accelerated browser. I do not have the --incognito option set.

KenT2, 2022-08-31:
PP uses a package called Selenium to control the chrome browser from Python in the file pp_chromeplayer.py. PP itself does not set any cookies but I don't see why cookies set by manual clicking should not be stored so they will take effect when the browser is run in PP.
If you set the cookies using the browser manually in advance of using PP do they get stored? Maybe when controlled by Selenium cookies set manually are not excepted.

PublicBenemy, 2022-08-31:
Thank you very much for your immediate reply! I just learnt that my issue is not cookie related, but that the settings of the site I'm loading are stored in a "Local Storage" space. I will investigate further and try what I can achieve. Thanks for pointing me to the pp_chromeplayer.py, that already helps. I will post it here when I find a solution.

closed 2023-02-25 by KenT2