KenT2/pipresents-beep #57: typing code on keyboard

AlainCx, 2022-09-16:
Hello,
for a futur project I'd like to use keypass to read audio files.
The visitors could use a small numeric keyboard to enter codes which could play audio files.
"2012" plays sound 1
"3615" plays sound 2
etc

Is there a way to do such a thing ?

Thanks.

Alain

KenT2, 2022-09-18:
Yes
If your keypad plugs into a USB port and will act as a numeric keyboard then you can use the pp_kbddriver_plus I/O plugin

If you keyboard has a RS232 interface then you can use the pp_serialdriver I/O plugin. You will need a RS232 to USB convertor.

Both allow you to define strings of characters which will generate input events.

AlainCx, 2022-09-18:
Tanks for the answer.

Which type of projet allows to use strings in Pipresents ?

Regards,
Alain

KenT2, 2022-09-18:
any type of show but only the two types of I/O plugin I described. You could write your own I/O plugin

AlainCx, 2022-09-18:
OK, I'll test this.

Thanks again !

closed 2023-02-25 by KenT2