KenT2/pipresents-beep #61: error: incorrect number of fields in control:

ecord17, 2023-02-08:
I read and added the information to the screen.cfg file for the touch screen. When I run the show, I am receiving this error. I added the same number of controls to the Main control tab and still receiving the same error

KenT2, 2023-02-08:
something is wrong in a Controls field not screen.cfg.
If you validate the profile using the editor it should give you more information.

ecord17, 2023-02-08:
awesome that did it.  I deleted all extra videos so that i make sure that I do it correctly. Right now it only showing one video.  Here is a screen shot. From what I read: I am getting an error "unknown command in control: pp-play0 vid0.  I wanted to match what was exactly in screen.cfg. I think maybe I read it wrong.  I am sorry
[image]

by the way, I did contribute on your website.

ecord17, 2023-02-08:
[image]

closed 2023-02-08 by ecord17