KenT2/pipresents-beep #64: Raspberry Pi Desktop for PC

PiSignage, 2023-03-08:
Hi, is there a way to run pipresents on Raspberry Pi Desktop for PC?

KenT2, 2023-03-08:
I haven't tried it but I suspect some of will run.

The Raspberry Pi desktop is just the Raspberry Pi OS, hence Linux, hosted on a different processor with a different GPU and I/O hardware.

Pi Presents will probably run but there may be a few minor things to sort out with initialising displays and I/O
GPIO will not work but anything USB related probably will
Video tracks using omxplayer will not work but VLC video tracks will probably work
Image tracks and audio tracks will probably work.

PiSignage, 2024-03-16:
It been a long time since I post this question. From my testing it will not start up as it look for omxplayer that can't be installed.

I have also tested the version above this one will not start I can't install python-pil or python-pil.imagetk

And with latest version kms will not start