KenT2/pipresents-beep #65: Add a gpio button to print the displayed file

dominiquegrota, 2023-04-12:
Hi, 
I thank you for your very good app ! I love to use it. 
It is not really an issue, it is more a enhancement... I would like to propose to print the shown picture when print on a gpio button. 
Is there a way to execute part or all of this ? :
- create a new "pp-printer" button linked to gpio (I suppose in /pipresents/pp_ressources/pp_templates/gpio.cfg ?)
- get the picture file name when the button is pressed
- light a gpio led to show the printing is fired 
- execute shell comamand "lp picture.jpg" to the default printer.
Thank-you for your answer...
Dominique

KenT2, 2023-04-14:
You need to write a track plugin. There is an example in the pp_camera_1p4 example that takes a picture with the camera and displays it. This will  give you some idea on how to proceed.  Also a section of the manual with examples.

track plugin configuration files will enable you to specify the file name to be printed.

You may need to run the print command as a separate sub-process as PP uses cooperative scheduling and no bit of code should run/block for more than a few hundred milliseconds.

dominiquegrota, 2023-05-04:
Thank's for your answher. It helped me to understand how pipresents works.... I tried dozen's of hours to understand how plugin and showhs work, but ...
I don't know if I do something wrong, but I could not get work a liveshow whit a track plugin. 
I actually copy pictures to display in a liveshow, and I would like to execute a script to print the displayed file in liveshow. I could not get the command stop the show to execute the plugin with the filename as attribute...