KenT2/pipresents-beep #66: Issue starting a show

Jgawkub, 2023-08-17:
Hi, 

I'm having an issue when trying to run one of the downloaded examples. Any time I attempt to run I get the following 

python3 /home/pi/pipresents/pipresents.py -p pp_mediashow_1p4
Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 968, in <module>
    pp = PiPresents()
  File "/home/pi/pipresents/pipresents.py", line 354, in __init__
    self.vp=VibePlayer()
  File "/home/pi/pipresents/pp_vibeplayer.py", line 14, in __init__
    self.drv2605 = DRV2605()
  File "/home/pi/pipresents/DRV2605_lib.py", line 265, in __init__
    BitField('period', 0xFF),       # period (us) = period * 98.46
  File "/home/pi/.local/lib/python3.7/site-packages/i2cdevice/__init__.py", line 166, in __init__
    self._i2c = smbus.SMBus(1)
FileNotFoundError: [Errno 2] No such file or directory

Installed on a clean version of the legacy OS and believe I followed all the install instructions properly. Any help appreciated!

KenT2, 2023-08-17:
Possibly a bug but could you check that you did this:

python3 -m pip install DRV2605

The Readme is not well formatted and you may have missed it. If not report back and I will try a clean install.

Jgawkub, 2023-08-22:
Hello, pardon the delay. I ran that command and here is what came out. 

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: DRV2605 in ./.local/lib/python3.7/site-packages (0.0.3)
Requirement already satisfied: i2cdevice>=0.0.6 in ./.local/lib/python3.7/site-packages (from DRV2605) (0.0.7)
Requirement already satisfied: smbus in ./.local/lib/python3.7/site-packages (from DRV2605) (1.1.post2)

Trying to run a profile I got the same error. 

Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 968, in <module>
    pp = PiPresents()
  File "/home/pi/pipresents/pipresents.py", line 354, in __init__
    self.vp=VibePlayer()
  File "/home/pi/pipresents/pp_vibeplayer.py", line 14, in __init__
    self.drv2605 = DRV2605()
  File "/home/pi/pipresents/DRV2605_lib.py", line 265, in __init__
    BitField('period', 0xFF),       # period (us) = period * 98.46
  File "/home/pi/.local/lib/python3.7/site-packages/i2cdevice/__init__.py", line 166, in __init__
    self._i2c = smbus.SMBus(1)

KenT2, 2023-08-22:
OK, I'll try a clean install.

KenT2, 2023-08-26:
Found the problem. You need to enable the I2C interface in Raspberry Pi Configuration>Interfaces.

This is a workaround. I need to change the code so you do not need to switch on I2C unless you need it.

Jgawkub, 2023-08-27:
That worked. Thank you for your help!

KenT2, 2023-09-03:
Issue now fixed in latest version. i2C is required only when the --vibes command line option is included or there is a DRV2605 I/O config file.

closed 2023-09-03 by KenT2