KenT2/pipresents-beep #67: Show timeout in subshow

AlainCx, 2023-09-06:
I'm working on a project consisting in a main radiobuttonshow and many other radiobuttonshows as subwhows. Several levels like russian dolls.
Is it possible to set up a global timer for each subshow to come back to the main subshow ?
I tried to use "show timeout" but each time it brings to the pevious show, not the main and first one.

Thanks !

AlainCx, 2023-09-13:
Ok, understood. Hyperlinkshow did the job in one case. Thanks !

One more question : In a Radiobuttonshow, I'm trying to use "Show Control at end" commands. I want to go to a specific show after an other show ends (using show timeout duration). It works, but commands seem disabled even if "Controls in subshows" are enabled. 
Do you know why ?
What are the best commands to use between "open", "openexclusive"... ?

Thanks again !

KenT2, 2023-09-17:
Not sure what you mean. You say it works but also commands are disabled

Open Exclusive is useful if PP does not know which shows are currently running.

AlainCx, 2023-09-18:
I tried "openexclusive"command but when the the desired show is opened, command are disabled for this show. Even if "Controls in subshows" is checked.
Any idea why ?

KenT2, 2023-09-19:
The problem might be that Open or OpenExclusive opens the Show as a top level show, not as a subshow.

Hyperlinkshows have jump or  goto commands which will work because they go to tracks in the same show.