KenT2/pipresents-beep #68: VLC Player

ErinC13, 2023-09-19:
I'm new to Raspberry Pi and I'm trying to follow your instructions on downloading & running PiPresents Beep. When I run the      python3 /home/pi/pipresents/pipresents.py, I get this message
alak13@vorhees13:~ $      python3 /home/valak13/pipresents/pipresents.py
Traceback (most recent call last):
  File "/home/valak13/pipresents/pipresents.py", line 981, in <module>
    pp = PiPresents()
  File "/home/valak13/pipresents/pipresents.py", line 135, in __init__
    self.mon.log(self,'\n'+ check_output(["omxplayer", "-v"],universal_newlines=True))
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1823, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'omxplayer'

I've read that OMX Player has been deprecated and that the closest to it is VLC. How do I change this so it access the VLC player instead of OMX Player? I just want to create an interactive singing pumpkins prop for Halloween. Thanks!

KenT2, 2023-09-20:
Are you running RPi OS legacy (Buster). When I installed Buster a week ago it still had OMXPlayer?
also Beep works only with the pi username. I suspect you have Bullseye.

There is a new version Pi Presents-KMS which works with Bullseye and with usernames other than pi.  Github says this is a development version but I have just finished squashing the last bugs and it is now ready for release to the world.

It would be great if you could try that one. It uses MPV instead of VLC as the video player. Audio tracks uses mplayer

ErinC13, 2023-09-20:
Thanks for such a quick reply! I'll try KMS today. Hopefully that'll work!

ErinC13, 2023-09-20:
I was able to get KMS to work. The only issue I'm having now is figuring out how to make the steps the tutorial from the DIY instructions work. 

I'm using the instructions from this site https://www.diymachines.co.uk/interactive-performing-pumpkins. I followed your instructions on how to install KMS, but am trying to follow his instructions to download the media files. His instructions say to use this command: python /home/pi/pipresents/pipresents.py -p DIYM -f -b. I tried and it doesn't work. I noticed that it's not in the same format as what's in your instructions. I've tried making it match the best that I can to the format yours is in, but I'm not  having any luck making it work. I'm super new to all of this; this in the 1st thing I've ever tried doing with Raspberry Pi so this is all a little foreign to me. Any help is super appreciated!

KenT2, 2023-09-23:
I downloaded the profile to see what the problem was since you did not tell me. What was required was to upgrade the profile for the later version of Pi Presents. I have done that for you and the it, minus the media is attached. You will need to edit the filename to remove.txt you can then extract it
The command line is OK.

[DIYM.zip.txt](https://github.com/KenT2/pipresents-beep/files/12705921/DIYM.zip.txt)